techniq/node-pn532 #4: Add support for reading/writing to Mifare Classic tags

techniq, 2015-01-10:
Library currently supports NTAG* tags, and hopefully other 4-byte page tags (Mifare Ultralight), but it currently does not support Mifare Classic tags, which have 16-byte blocks.

yangpu, 2015-12-24:
Do you have any plan to work on Mifare Classic tags?

I have some NFC tags(ISO14443A), and door access card and bus card (I don't know the protocol and type), all tags uid can be detected. However, I'd like to write some data into NFC tags, but nothing is response.

Looks like this repository(https://github.com/tessel/rfid-pn532) can surrport Mifare Classic  block read/write. However, its hardware is more specified to Tessel.

techniq, 2015-12-24:
I have plans to support Mifare Classic tags, but not in the immediate future.  I think I ran into issues getting the authentication working.  I [started](https://github.com/techniq/node-pn532/blob/master/src/pn532.js#L338-L374) support for this, but couldn't get it to work at the time.  I actually used the tessel library (along with a few others) when I was trying to understand the data flow.  I even had an [issue](https://github.com/tessel/rfid-pn532/issues/17) open with them about merging our efforts a while back.

I believe most everything is in place except the auth side of things.  There might also need to be some changes/configuration when reading as some cards/tags have different block/page sizes.  I have a spreadsheet I used to document my findings you might find helpful - https://docs.google.com/spreadsheets/d/1IJNzGNl7qdgCePklrghU54n4hOx5E6y20rEpZw0gJhc/edit#gid=1815263715

PRs are always welcome.  Feel free to compare my `authenticateBlock`, `readBlock`, `readNdefData`, `writeBlock`, and `writeNdefData` and compare to (tessel's)[https://github.com/tessel/rfid-pn532/blob/master/index.js#L530-L643]