techniq/node-pn532 #8: Remove listeners and scanTag timeout

alexit84, 2016-01-23:
Hello,
I'm playing with your library with Raspberry Pi and PN532 Breakout board. Everything is working well :)

I had the need to set a timeout to the scanTag() function if if nothing had been found. So I needed a function to stop all the pending listeners.
This worked like a charm for me:

```javascript
removeListeners() {
        this.frameEmitter.removeAllListeners();
};
```
Maybe this can be useful for others.

Anyway, maybe a timeout parameter on the scanTag() function could be the best option...

Thank you,
Alessandro

techniq, 2016-01-27:
Glad to hear it's working well :)

Not sure I fully understanding the need to remove the listeners.  Are you calling `scanTag` directly (and then waiting on the promise), or listening for the `tag` event (which calls/polls `scanTag`)?

Thus far I typically just listen for the [tag](https://github.com/techniq/node-pn532/blob/master/examples/card_scan.js) event and respond whenever that may be (seconds/minutes/days...).

alexit84, 2016-01-27:
Hi,
you are right, I've not explained well my need. 
I'm making a simple REST server with Express to get data from Nfc tags. I call directly the scanTag waiting on the promise, but I would like to stop that request after a set timeout if an event didn't happen.
This is an example of my code:

```javascript
app.post('/read', function (req, res) {
    var timeout = setTimeout(function () {
      res.status(504).send("No tag found");
      rfid.removeListeners();
    }, 10000);

    rfid.scanTag().then(function(tag) {
        clearTimeout(timeout);
        console.log("Tag: "+ tag.uid);
        var result = {};
        result.uid = tag.uid.replace(/:/g,'');
        rfid.readNdefData().then(function(data) {
            if (data != null || data != undefined) {
                var records = ndef.decodeMessage(data.toJSON().data);
                if (records[0].type == "U")
                    result.uri = records[0].value;
                else if (records[0].type == "T")
                    result.text = records[0].value;
            } 
            console.log(JSON.stringify(result));
            res.status(200).send(result);
        });
    });
});
```
If the result does not come within 10s, the user is informed and the promise is cancelled (he can not wait indefinitely...)  
I'm happy with the result I got, but probably there are better ways to accomplish the same results. I'm opened to suggestions :)

Thank you,
Alessandro

techniq, 2016-01-27:
Awe, makes sense know seeing your use case.  I could see having the ability to timeout or cancel a command.  I think currently you could access the `frameEmitter` directly without having to patch in the direct `removeListeners()` 

```js
...
    var timeout = setTimeout(function () {
      res.status(504).send("No tag found");
      rfid.frameEmitter.removeListeners();
    }, 10000);
...
```

Since promises can't be cancelled, it might be worth looking at using Observables/[RxJS](https://github.com/ReactiveX/RxJS) which allow cancelling a "future" value

alexit84, 2016-01-28:
Unfortunately that line give this error:
`TypeError: rfid.frameEmitter.removeListeners is not a function`

kelvien, 2016-04-15:
@techniq Hi, what is the proper way to remove rfid's ready listeners? 
I tried deleting the rfid object, but it seems like the "ready" listeners is still running. I've tried using "Event-Emitter"'s .off function, but it doesn't work the way I expect it to.