techniq/node-pn532 #11: Read data example not decoding NDEF records

kmoe, 2016-05-23:
Using the read_data.js example, I can't get it to show the parsed NDEF records. It correctly displays the buffer data, but the [next line](https://github.com/techniq/node-pn532/blob/master/examples/read_data.js#L19) seems to fail silently.

I've tried with NTAG216 and various Mifare cards.

Here's the log:
```
@katy ➜  node-pn532 git:(master) ❤ node examples/read_data.js
Waiting for rfid ready event...
Listening for a tag scan...
Tag { ATQA: <Buffer 00 44>, SAK: 0, uid: '04:98:e8:12:ff:38:80' }
Reading tag data...
Tag data: <Buffer d1 01 0f 54 02 65 6e 48 65 6c 6c 6f 20 43 61 72 6c 6f 73>
```

closed 2016-05-24 by techniq

techniq, 2016-05-24:
Took me a longer than hoped this evening as I was having some USB UART cable issues, but I was finally able to take a look at this, and after some digging, it appears the structure I used to send to `ndef.decodeMessage(...)`, `buffer.toJSON()` is no longer correct, and was silently raising an error within `ndef.decodeMessage`.  This might be from a breaking change from Node v4 as I don't think I've tested this since I went from 0.12 to 4.0 to remove the need to transpile the source.

I've updated the README and `read_data.js` example, but you should just need to simply change `data.toJSON()` to `Array.from(data)` if you are running the example.  Let me know how it goes.

[image]

kmoe, 2016-05-24:
Hey @techniq, thanks so much for the quick response. By using `Array.from(data)` I've managed to read data from a tag I'd written from an Android app, but I'm getting errors reading from a tag I wrote to using your 'write' example. Would you mind double checking that one as well to make sure it's still working for you?

techniq, 2016-05-24:
I wrote my tag using am Android app as well. Getting ready to call it a
night, but I'll try to take a look tomorrow night if I get some time.

kmoe, 2016-05-24:
No problem. Thanks for your help. I've just discovered that for some reason, writing only one NDEF record causes the problem, but if I have two NDEF records it's fine. I'm happy with that workaround.

techniq, 2016-05-24:
It looks like `encodeMessage` [requires](https://github.com/don/ndef-js/blob/master/lib/ndef.js#L223) an array to be past.  When writing 1 record, were you still wrapping it in an array?

kmoe, 2016-05-24:
Yes, I was still passing an array.

techniq, 2016-05-24:
If you could send me the logs with `PN532_LOGGING=debug ...` turned on and write `ndef.encodeMessage([ndef.textRecord('test')])` I can try to dig into the issue.  I have a [spreadsheet](https://docs.google.com/spreadsheets/d/1IJNzGNl7qdgCePklrghU54n4hOx5E6y20rEpZw0gJhc/edit#gid=0) I wrote a while back that might help re-jog my memory on this stuff :)

commit dc4a6811 (2016-05-24): Fix read data example to work with Node 4.0.  Fixes #11