techniq/node-pn532 #14: i2c problem

Stan92, 2017-05-22:
Hi,
I tried your package, and it seems it doesn't work in i2c mode.
The problem comes from this line 
var rfid = new pn532.PN532(wire);
I get this error

```

/home/pi/rfidtest/node_modules/i2c/lib/i2c.coffee:88
          return callback(err);
                 ^
TypeError: callback is not a function
```

Do you have any idea?
thanks

techniq, 2017-05-22:
Hi @Stan92.  Sorry, but i2c in the released versions is not functional.  For my projects I've always used UART/Serial, although I did make some progress in the [i2c branch](https://github.com/techniq/node-pn532/tree/i2c/src).  Last I remembered I had it working successfully over Firmata but had some [hard coded pins](https://github.com/techniq/node-pn532/blob/i2c/src/pn532_i2c_firmata.js#L19-L20) and it still needed some work before I could merge it back to master (are the IRQ pins required or can I fall back to polling etc).

I've not done a lot of work with I2C myself (just using other's libraries, etc) but it seems I2C is more of a "pull" approach while UART is more "push" (UART you can wait for data to come to you, I2C you have to go get it, or at least that's where the IRQ pin on the pn532 comes in to let me know data is available to go fetch).

Anyways, you can try to experiment with the i2c branch, but you'll need to pull from source.  I'm also always willing to help others like I did with [card emulation](https://github.com/techniq/node-pn532/issues/12) and would love for someone to get this going and send a PR.  I've not had a project recently that used this library so it's not been a priority for me to work on.

Stan92, 2017-05-22:
No problem.. I thank you for explanation (I've learned something about pull vs push approach)..
Thanks once again.

closed 2017-05-22 by Stan92